SnijderC/dyplayer #1: Checking of the library with the DY-SV5W and DY-SV8F

ntapsis, 2020-01-26:
The library is very interesting, but it needs some corections. 

1. In the library I didn't find the follow commands from the manual:

Query Command
Command: Query current play drive
Command code: AA 0A 00 B4

Setting Commands
Command: Specified song to be interplay
Command code: AA 16 03 Drive S.N.H S.N.L SM

Command: Combination play setting
Command code: AA 1B length SM

Command: End Combination play
Command code: AA 1C 00 C6

2. In the README.MD, in the table with the modules, the first "DY-HV20T" has to become "DY-SV5W"

3. I checked the examples of the library with the modules DY-SV5W and DY-SV8F. None of the examples didnt work with my modules.
My settings were:
Arduino UNO v.3
Mode: UART (0-0-1)
Power supply: 5V
RX and TX connections through 1K resistors with Arduino's TX and RX.

Am I doing something wrong?

SnijderC, 2020-01-28:
Thanks for the extensive report, I will go through everything and reply point by point, perhaps in one stretch or maybe in several. First of all Query current pla drive seems to be missing indeed. Also based on this I found another bug in `DYPlayer::getDevice()` that I made an issue for #2. To be continued.

SnijderC, 2020-01-28:
> In the library I didn't find the follow commands from the manual:
> Query Command
> Command: Query current play drive
> Command code: AA 0A 00 B4

It was not my goal to make 100% coverage for all functionality, so this is indeed missing, I had no need for it. That said, I can add it: #3

ntapsis, 2020-01-28:
I would be glad to test the new version of the library on DY-SV5W, DY-SV8F and DY-HV20T which I owe. Thank you for your efforts

SnijderC, 2020-01-29:
> I would be glad to test the new version of the library on DY-SV5W, DY-SV8F and DY-HV20T which I owe. Thank you for your efforts

That would be very helpful, I'm going to probably order some of these boards soonish too so I can test better but don't have them yet. 

As you can see I didn't get through all of your feedback yet, but will get back to it. It's easier for me to test with ESP32 boards, since I have plenty and I find them convenient to use. Plus I want things independent from the Arduino framework. So I added support for that in a feature [branch](https://github.com/SnijderC/dyplayer/tree/esp32-hal) (WIP).

Also.. Thank you for your efforts too!

ntapsis, 2020-01-29:
I respect your time, so I try to be precise. Please check if there is a problem to the follow

Missing functions (maybe to add):
1. Command: Check Current Playing Device
Command Code: AA 0A 00 B4
Return: AA 0A 01, drive, SM

2. Command: Combination play setting
Command Code: AA 1B Length H-L H-L SM

3. Command: End Combination play
Command Code: AA 1C 00 C6

Maybe incorrect code (maybe to change):
1. Command: Select specified file to interlude
Command Code: AA 0B 03 00 00 00
Instead of Command Code: AA 16 03 Drive S.N.H S.N.L SM
Line in the DYPlayer.cpp: 220

2. Command: Set Cycle times
Command Code: AA 19 01 00
Instead of Command Code: AA 19 02 H L SM
Line in the DYPlayer.cpp: 242

3. Command: Select but no play
Command Code: AA 1F 2 00 00
Instead of Command Code: AA 1F 02 S.N.H S.N.L SM
Line in the DYPlayer.cpp: 254

SnijderC, 2020-01-29:
The missing commands now have issues.

The maybe incorrect commands I am going to check first, soon.

> Connections RX/TX through 1K resistors.

I added a note on this in the README because the manual says you should use 10K resistors..

> First I made the connections and I tried to upload the sketh to Arduino, but I was receiving error, as there was no connection between Arduino and PC. The problem solved when I disconnected the UART cables.

Uno only has 1 UART, it's shared between the project and uploading sketches, so that's expected. Maybe I can add a line or two in the readme about that.

Also I noticed I swapped CON1 and CON3 in the README by accident, but since you got sound I guess that didn't stop you from wiring it up.

>In the examples the Volume was set to 15, which coresponds to 21 out of 30. So it is not 50%, but 70%.

Where did you find this information? I assumed it would be an integer between 1 and 30, I guess it is not one-on-one then?

> Functions byPathCommand don't work at all. I think that it has to do with the char path[]

I think I did see that working but the path is very specific, I'll test that again.

I am not sure what is wrong with the interlude command, it is yellow in the first column, any reason?

DYPlayer::setCycleTimes and DYPlayer::select are probably fixed, but untested.

ntapsis, 2020-01-30:
> Where did you find this information?

-In UART mode user's guide at page 2 says "All data is HEX in protocol, unless special introduction".

> what is wrong with the interlude command
The command is written as "AA 0B 03 00 00 00", but I think that it has to be "AA 16 03 Drive S.N.H S.N.L SM".

I organized better the 29 functions
[DY-XXX.pdf](https://github.com/SnijderC/dyplayer/files/4131816/DY-XXX.pdf)

About the path maybe there is an other way
[path.pdf](https://github.com/SnijderC/dyplayer/files/4131817/path.pdf)
Are there any rules about the length of the name of the folders and the files?

Maybe instead to define the path, it would be easier to do the same with the "Combination play" where the user defines an order of the sound files to play.
The user could declare at the begining of the sketch which is the first and which the last file in each folder.
combinationPlay(uint16_t number, uint16_t number)
--

SnijderC, 2020-01-30:
> -In UART mode user's guide at page 2 says "All data is HEX in protocol, unless special introduction".

Ah I see what you mean now, but I send a byte containing an integer of `15` that would send `0x0f`.. not `0x15`. The idea is that as a user of the library you don't have think thing about such things.

> I organized better the 29 functions

Nice I think there should be an overview like that in the repo at some point. But also the API explained.

> About the path maybe there is an other way
> Are there any rules about the length of the name of the folders and the files?

I am not sure what kind of problem you are trying to solve here. I have found some PDFs in a link on youtube that contain more info: https://yadi.sk/d/PA3xDmBEXSVtpA Read the `uart mode user's guide.pdf` at the end there is something on paths. I think you can make a path any way you like except for the combination play feature. I was able to play by path as in the example IIRC. 

```
#include <Arduino.h>
#include "DYPlayerArduino.h"

DY::Player player;
void setup() {
  player.begin();
}
void loop() {
  char path[] = "/00001.MP3";
  player.playSpecifiedDevicePath(DY::FLASH, path);
  delay(5000);
}
```
Note this is from the latest version on the master branch, not yet on platformio (if you use that). I added a namespace, you will need to adjust your sketches, basically add `DY::` to anything from the library (as above).

> Or maybe you could be inspired from the command list of the WT5001

Can't really add functionality to the module that isn't there eh.. :)

Just give me the weekend to figure out some stuff and then you can test everything and tell me if anything is still broken.

ntapsis, 2020-01-31:
I checked how the several modes of play work with the hardware.

Conclusions:
1. Play. Plays the first song and stops. It doesn't continue to the next song, except if it's combined with the play mode.

2. Play Specified. The command uses the number of the order of the song. Not the exact name of the file. For example we put 01, 02, not 00001.mp3, not 00002.mp3. So playSpecified(orderInDir).

3. Play Specified Device Path. The command uses the exact name of the folder and the file. For example "/A/*00002*MP3".

4. Interlude Specified Device Path. The same as with the previous. I don't understand which is the diference with the previous.

5. Play combination. I tried all the possible ways with the UART tool and I didn't make it to work. The UART tool accepts order of the songs and not names. The user's guide didn't help. The good with this function is that it can make a play list, something which I strongly need.

Have a nice weekend

ntapsis, 2020-02-02:
Sorry Chris, I am not programmer. I made it work in that way (except the combination play and the unknown commands)
[TestingDY.txt](https://github.com/SnijderC/dyplayer/files/4145327/TestingDY.txt)
You can do it much better

SnijderC, 2020-02-04:
> Conclusions:
> 
> 1. Play. Plays the first song and stops. It doesn't continue to the next song, except if it's combined with the play mode.
>    Edit: Problem solved when I **put play mode=06**

That's the default mode of the module. You can set it by calling `DY::DYPlayer::setCycleMode(DY::PLAY_MODE_SEQUENCE)`

> 2. Play Specified. The command uses the number of the order of the song. Not the exact name of the file. For example we put 01, 02, not 00001.mp3, not 00002.mp3. So playSpecified(orderInDir).

Yes, that's why the manual recommends you follow `00001.mp3` - `65535.mp3`, if they all have a name like that you can play them by number. This is also just how the module works. You could use `DY::DYPlayer::playSpecifiedDevicePath` instead to get around it, though that's slightly more data to send and more ROM used.

> 3. Play Specified Device Path. The command uses the exact name of the folder and the file. For example "/A*/00002_MP3".
>    Edit: The command works with the name of the folder too in the format: /[name of the folder]_/*MP3

It should work just with: `/A/00002.MP3`, because there is a bit of code that deals with the funky file format for you so you don't have to.

> 4. Interlude Specified Device Path. The same as with the previous. I don't understand which is the diference with the previous.

When a sound is playing, you can pause that sound and play a new sound until the second sound finshed or until you call `DY::DYPlayer::stopInterlude`, then the module will continue with the first sound. You could call it interrupting the sound for another if that's more clear.

> 5. Play combination. I tried all the possible ways with the UART tool and I didn't make it to work. The UART tool accepts order of the songs and not names. The user's guide didn't help. The good with this function is that it can make a play list, something which I strongly need.

I will have a look at this soon, I didn't find the time yet.

>    Edit: **Alternative solution** is to put the songs of each playlist in seperated folders and then to give command to play the songs of a specified folder

This is what I wanted to suggest to you :)

> Have a nice weekend

Thanks!

I have worked on the library in a feature branch last weekend but it's not in a state that it works, so I didn't push it to Github yet. I have added API documetation there, that might also help a bit with finding out what is what and how to use it. I also renamed some functions for consistency and some to better match the manual. Hopefully will push to Github before the end of the week.

ntapsis, 2020-02-08:
I checked the examples of the v.3 of the library with Arduino UNO and DY-SV5W.

Play All Sounds: ok
Play Sound by Number: ok
Play Random: ok

Play Sound By Path: I use the module with SD card, but arduino IDE returns 'FLASH' is not a member of 'DY', 'SD' is not a member of 'DY', 'USB' is not a member of 'DY', 'U' is not a member of 'DY'. So it is something with the declaration of the device.

Combination play: didn't work

SnijderC, 2020-02-08:
> Play Sound By Path: I use the module with SD card, but arduino IDE returns 'FLASH' is not a member of 'DY', 'SD' is not a member of 'DY', 'USB' is not a member of 'DY', 'U' is not a member of 'DY'. So it is something with the declaration of the device.

They should be a member of `DY::Device`, so: `DY::Device::FLASH`.

> Combination play: didn't work

Most probably you need to figure out what the directory name should be on your board. On mine the files had to be in a directory called: `ZH` but the manuals from different modules show different names, like: `DY`, `XY` and others.. I am not sure what works for your board.. But I tested combination play and it worked fine.

ntapsis, 2020-02-09:
> They should be a member of DY::Device, so: DY::Device::FLASH

Play Sound By Path worked with DY::Device::USB, but not with the DY::Device::SD
It worked only with the sounds directly in the directory and not with the files inside a folder.
I used my everyday USB flash which was containing several other files too (eg pdf, docs, etc) and it worked fine as an external storage drive.
About the combination play, all the music files were inside the DY folder. I checked with USB and SD without luck. Later I will check with other names too. For combination play in which drive the music can be?

SnijderC, 2020-03-08:
I just came back from a holiday, so didn't see this until now. Thanks for the update!

Looks like I can do some improvements, e.g. make all small letters capitals automatically, then you can just write file names with `.mp3` instead of `.MP3` which is a bit more intuitive. 

> Edit: In my USB flash I just renamed the DY folder to ZH and it worked. Then I renamed again to XY and it didn't work.

I'll add the `ZH` thing to the readme, IIRC mine also required the name to be `ZH` and my manual also said `DY`.. 

> Play Sound By Path worked with DY::Device::USB, but not with the DY::Device::SD

I ordered a board with an SD card slot a while back but it's stuck in China since a few weeks, probably due to the "issue" they are facing over there. Until I get it I'm afraid I can't test the combination play feature with `DY::Device::SD`.

> Something unexpected happened. While I was using my regular USB flash disk, I heard sounds which I had in the disk and they didn't follow any rules for the names of folders or names of the files. > The only I am sure, is that this sound was the oldest sound in that drive. So maybe the system recognizes the files according the age and not according the name...maybe!

I think you are right, I had a similar problem with files in a hidden directory (trash folder), I heard sounds that I had removed already, which was kind of a suprise until I figured that out :sweat_smile: I guess that should be added to the manual too.

Not sure when I'll get to it but I'll try to work on it this week.

SnijderC, 2020-03-29:
Master now converst lower case paths to upper case paths and several other changes, I still need to update the readme with our findings. Still no module to to test with. 

Can you maybe explain more about what you meant by: 

> Is it possible to add a simple example with one command with return? For example the name of the playing song.

I can make an example that uses a return but AFAIK, I can't tell you the file name of the playing sound, the module doesn't do that. I can get only the number.

closed 2020-07-08 by SnijderC

fritsjan, 2021-04-30:
Hi, I am trying to play songs inside a folder like /A/00102.MP3. But that does not seem to work.
It only works on playing inside the root...
made a temporary solution by naming the songs like 00101.MP3, 00102.MP3 and 00201.MP3 00202.MP3 and so on, but I need to count the number of songs belonging together, This can only be done inside a folder.
Is there a way to just select a folder from where to play and then just play by number?

Using a DY-SV5W and a 32gb card

mo-norant, 2021-06-01:
@fritsjan do you have an example

fritsjan, 2021-06-03:
`#include <Arduino.h>
#include "DYPlayerArduino.h"
//#include <SoftwareSerial.h>

// Initialise on software serial port.
//SoftwareSerial SoftSerial(10, 11);
DY::Player player(&Serial1);

void setup() {
  player.begin();
  // Also initiate the hardware serial port so we can use it for debug printing
  // to the console..
  Serial.begin(115200);
  while(!Serial);
  player.setVolume(20); // 50% Volume
  Serial.print("playing device: ");
  Serial.println((int16_t)player.getPlayingDevice());

  //this 
  Serial.print("Sounds: ");
  Serial.println((int16_t)player.getSoundCount());
  delay(500);
   
  //player.setCycleMode(DY::PlayMode::Repeat); // Play all and repeat.
  //player.play();
}

void loop() {

  //this plays a sound
  char path[] = "/00001.MP3";
  player.playSpecifiedDevicePath(DY::Device::Flash, path);
  Serial.print("Playing sound: ");
  Serial.println((int16_t)player.getPlayingSound());
  delay(5000);

  //this does not play, but file exists
  char path2[] = "/A/00001.MP3";
  player.playSpecifiedDevicePath(DY::Device::Flash, path2);
  Serial.print("Playing sound: ");
  Serial.println((int16_t)player.getPlayingSound());
  delay(5000);

  //this does not play, but file exists
  char path3[] = "/B/00001.MP3";
  player.playSpecifiedDevicePath(DY::Device::Flash, path3);
  Serial.print("Playing sound: ");
  Serial.println((int16_t)player.getPlayingSound());
  delay(5000);
  
}`

fritsjan, 2021-06-04:
I have found the bug!
In DYplayer.cpp:
void DYPlayer::byPathCommand

replace: sendCommand(_command, len + 4); 
to: sendCommand(_command, _len + 4); 

it was using the **len** variable which gets updated when using paths to **_len** so therefore paths did not work.

fritsjan, 2021-06-04:
#35