SnijderC/dyplayer #8: Add command: Combination play setting

SnijderC, 2020-01-29:
`AA 1B length High Byte Low Byte ..... High Byte Low Byte SM`

First `High Byte Low Byte` is the first sound, second `High Byte Low Byte` is the *nth* sound.

The manual is not clear on how many sounds may play at the same time but it does mention that the sound files should only be 2 characters long!

ntapsis, 2020-02-02:
The combination play is difficult subject, because there is not enough information in the datasheets, or it is confusing. So I checked what happens with similar products.

According to the manuals, DY-SV17F and XY-V17B are almost identical.
Datasheet: http://attach01.oss-us-west-1.aliyuncs.com/IC/Datasheet/13288.pdf
https://github.com/electron1979/Arduino_XY-V17B_mp3_wav_player

The DY-SV17F manual (p.3) says:
It is combined by **file name**, **the file is stored in DY folder**， the file is named as two bytes , the number name is recommended such as 01.mp3, 02.mp3, and it is also named by two letters.
CMD： AA 1B length High Byte Low Byte ..... High Byte Low Byte SM
For example： AA 1B 04 30 31 30 32 8C. Set **the two files named ”01”, ”02”** to Combination play
-Q: What files he means? Music files or a playlist?

The XY-V17B manual (p.6) says:
"combination play is combined by **filename**. **The file requirements are stored under the "XY" file**. You can change the name of the file you want to combine to two bytes, which is generally recommended as a number. Such as: 01. Mp3, 02. Mp3."
-Q: Again he speaks about a file....

The WT001 manual (p.10) says:
"The combination of playing is sending 10 groups or less music combination code to the WT5001-48L continuously, WT5001-48L play the music according to the sequence of code received. Different from sending name to control directly is that the next code can not interrupt the playing until finish the current song , receive the command to do FIFO processing. Example : WT5001 continuously receive "7E 04 A8 00 08 7E ","7E 04 A8 00 06 7E ", "7E A8 04 00 07 7E "," 7E A8 04 00 04 7E "," 7E 04 A8 00 03 7E ","7E 04 A8 00 02 7E " six sets of data, WT2801-S specify to play SD files named" 0008.mp3 "," 0006.mp3 "," 0007.mp3 "," 0004.mp3 ","0003.mp3 "," 0002.mp3 "6 audio files in order."

In the manual of WT5001 (p.7) says too:
"There is an **iSound.mp3 document in SD card**" ..... "you can open iSound.mp3 file on the computer by "WordPad", modify the setting parameters inside".
"ISound.mp3 files must put at the final storing address of SD card or U disk, that is the final to deposit to the SD card or U disk ( the final of index sequence)."
https://github.com/valeriocieffe/Arduino_wt5001_48L

So....
Maybe for the combination play it is necessary a text file named 01.mp3 (which will contains a list of songs - a playlist), which is stored in the folder named "DY" in the flash or SD card (/DY/01.mp3). For example something like this https://github.com/NachtRaveVL/BY8X01-16P-Arduino/blob/master/README.md

Is it possible something like that?

SnijderC, 2020-02-05:
I added combination play (untested, no time right now). It's in the `add-combination-play` branch. I will test it soon, hopefully tomorrow.

SnijderC, 2020-02-05:
I tested combination play and got it working after some more tweaks. There is an example sketch for Arduino.

closed 2020-02-05 by SnijderC

SnijderC, 2020-02-06:
BTW, I had to put the files into a directory called `ZH`, I added remarkt in the readme, yours might be different..