SnijderC/dyplayer #12: More examples?

daPoppa, 2020-02-24:
This library looks promising for the ardunio project I've just started working on (I'm new to arduino), as I have an MP3 board with an chipset that should be compatible with this. 

Do you have any other examples of how to use the various commands?  eg:  checkPlayState,  how is this actually called and used in an arduino ini ? 

cheers

daPoppa, 2020-02-24:
trying to set up the RX/TX on pins 10/11, how does that syntax work?  

eg:  do I do something like this?   This results in an error.   Sorry I'm not understanding the method.

// soundboard pins and setup
SoftwareSerial Serial2(10, 11);   //RX and TX
DY::Player player(&Serial2);

daPoppa, 2020-03-08:
I guess this thread is not being monitored any longer?   

As an update, I did get my card working, playing the songs I want.  (I had to make sure to copy the files to the SD card in the exact order they are named)

However, the checkPlayState is returning a FAIL - any hints?

daPoppa, 2020-03-08:
The TX port on the uno works fine, but the RX port is returning FAIL.   How would I go about using different software serial pins with this library?  eg:  pin 11=TX, pin 10=RX ?

SnijderC, 2020-03-08:
Hi, I just came back from a holiday that's why I didn't reply until now. 

I'm not sure how SoftwareSerial should work, but I can have a look at that. I'll add a checkState example as well and check that it works as it should while I'm at it.

> trying to set up the RX/TX on pins 10/11, how does that syntax work?
>
> eg: do I do something like this? This results in an error. Sorry I'm not understanding the method.
>
> // soundboard pins and setup
> SoftwareSerial Serial2(10, 11); //RX and TX
> DY::Player player(&Serial2);

That looks okay to me, what kind of error did you get? A compile time error or a runtime error? In the intialization or when using the serial port? What board are you using?

> As an update, I did get my card working, playing the songs I want. (I had to make sure to copy the files to the SD card in the exact order they are named)

Yes that sort of thing happens because of the way the board is implemented. There are some tips about that in the readme.

> However, the checkPlayState is returning a FAIL - any hints?

Just to be clear, which output did you expect, e.g. did you check the play state while playing or when paused, etc.

> The TX port on the uno works fine, but the RX port is returning FAIL. How would I go about using different software serial pins with this library? eg: pin 11=TX, pin 10=RX ?

You mean you can control the board but you are not receiving anything from the board?

I'm not sure when I'll get time to check this out but I'll try to have a look this week.

daPoppa, 2020-03-09:
Thanks for the reply!   Hope you had fun on your holiday.
Thanks for the library as well.    When I try to use different PINS for the RX/TX,  using this command:

> // soundboard pins and setup
> SoftwareSerial Serial2(10, 11); //RX and TX
> DY::Player player(&Serial2);
> 

I get the following errors:

> In file included from C:\Users\Harry\Documents\Arduino\sounds2\sounds2.ino:3:0:
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:9:7: note: candidate: DY::Player::Player(HardwareSerial*)
>        Player(HardwareSerial* port);
>        ^~~~~~
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:9:7: note:   no known conversion for argument 1 from 'SoftwareSerial*' to 'HardwareSerial*'
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:8:7: note: candidate: DY::Player::Player()
>        Player();
>        ^~~~~~
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:8:7: note:   candidate expects 0 arguments, 1 provided
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:5:9: note: candidate: constexpr DY::Player::Player(const DY::Player&)
>    class Player: public DYPlayer {
>          ^~~~~~
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:5:9: note:   no known conversion for argument 1 from 'SoftwareSerial*' to 'const DY::Player&'
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:5:9: note: candidate: constexpr DY::Player::Player(DY::Player&&)
> C:\Users\Harry\Documents\Arduino\libraries\dyplayer/DYPlayerArduino.h:5:9: note:   no known conversion for argument 1 from 'SoftwareSerial*' to 'DY::Player&&'
> sounds2:13:12: error: redefinition of 'DY::Player player'
>  DY::Player player;
>             ^~~~~~
> C:\Users\Harry\Documents\Arduino\sounds2\sounds2.ino:10:12: note: 'DY::Player player' previously declared here
>  DY::Player player(&Serial2);
>             ^~~~~~
> exit status 1
> no matching function for call to 'DY::Player::Player(SoftwareSerial*)'
> 

If I just initialize the player using the default serial ports, it works for the TX (out of the arduino to the RX of the music card).   The arduino RX is connected to the TX of the sound card.  I've run the scripts connected to the computer and also only running with a separate power supply. 

> DY::Player player;     

I've used several of the commands, and the ones I'm planning on using are:

> player.stop();
> player.setVolume(nn);
> player.playSpecified(nnnn);

However, when I try the checkPlayState function, it returns DY::PlayState::FAIL, instead of DY::PlayState::PLAYING, etc.
If I try the DY::DYPlayer::getPlayingSound, it always returns a 0, regardless of whether a sound is playing or not. 
This seems to indicate the the communication between the card and the arduino is not functioning using the default serial ports/pins.

I have implemented a work-around, by using the BUSY pin on the card, which returns a 0 if a song is playing OR ~3v if idle.  The issue with that however, is that the BUSY pins takes about 900 ms to switch states between idle->play, or play->idle (which can be handled in code).

cheers

SnijderC, 2020-03-28:
I will have a look at this but in the mean time, make sure the RX is not still connected to the UART chip that is connected to the computer. This is important because the RX line can only work correctly if there is only 1 device attached. You would get probably get `DY::PlayState::FAIL` if the RX line is not working correctly, I think it's the most probable reason to get that return value.

daPoppa, 2020-03-28:
thnx for checking into this.  Yes, I disconnected the Arduino from the computer when trying this out, but still got the fail.   I have been playing around with the BUSY pin and that works OK, although it does take about 900ms to change state, which I can work around.
I'll play around with the RX some more today. 
cheers

_________________________________________________________________

All we have to decide is what to do with the time that is given to us.

ghoeffner, 2020-03-30:
Seems like it is not compatible with SoftwareSerial yet?
At least I couldn't find anything related to swserial in the source.

daPoppa, 2020-03-30:
OK thnx, that makes sense.  The BUSY pin is working for me, so I'll just keep going down that route. 
cheers

_________________________________________________________________

All we have to decide is what to do with the time that is given to us.

SnijderC, 2020-03-30:
I have checked out software serial and there was something obviously not implemented on my part to support that. It was not very hard to fix though. I added support for it and made an example file that fetches data from the module as well, i.e. your use case. I tested it on a Arduino Nano clone on pin 10 and 11, the same pins you use. I get valid responses from the module. The code is in master, will do a new release to platformio soon too.

Please check your board and the "Limitations" chapter on this page to make sure the board can do interrupts on the pins you selected. https://www.arduino.cc/en/Reference/SoftwareSerial

SnijderC, 2020-03-30:
Please let me know if it works so I can close this issue.

ghoeffner, 2020-03-31:
swserial works absolutely flawless now. Great work.

closed 2020-03-31 by SnijderC

daPoppa, 2020-04-01:
I've installed the new library, but it's still not working as expected.   I'm using an arduino uno, pins 10/11 (RX/TX) and the sound board DY-SV5W.  The arduino TX (pin 11) to RX on the SV5W works fine, sounds all play as specified, but in the serial monitor, it continually outputs "Playing sound: 0", which seems to indicate that the call is not succeeding.  The example code I used was the new one available here.  (SoftwareSerial example)
Any idea how I troubleshoot this?
thnx

SnijderC, 2020-04-01:
The only time I experienced this, is when a second device is attached to the TX line of the module,e.g. you connect the USB cable of an Uno to the computer or a FTDI board to the computer and an Arduino nano board and use the same serial port for communication wiith the module. 

Receiving data from the module then only works if the usb cable or ftdi cable is disconnected from the arduino board.

What might help me make more useful suggestions is if you share the type of Arduino board you are using and how you connected the module to it, perhaps a photo could help. That way I can see what the supply voltage of the Arduino board is and check the connections to the module (resistors should be placed on the RX and TX lines between the boards if the supply voltage is > 3.3V. They should be 1KOhm. 

Further I can suggest you to attach an oscilloscope if you have access to one and see if you can detect data being sent on the line.

Lastly as I suggested before please check your board and the "Limitations" chapter on this page to make sure the board can do interrupts on the pins you selected. https://www.arduino.cc/en/Reference/SoftwareSerial

daPoppa, 2020-04-01:
Thnx for the hints. 
Yes, I had the usb cable connected, which I thought made sense, since I was using different pins 10/11 and the scripts says to use the monitor window for debugging.  How could I use debugging if the usb cable is disconnected?  
I'm using an Aduino Uno (knockoff), connected to pins 10/11, using the same script that you provided (thank you).  I have 1k resistors between both TX/RX - using the USB to power the arduino atm.  
Sorry, no access to an oscilloscope. 
Yes, this board appears to allow PWM~ on pins 10/11.  I have used those pins before with neopixels. 

#include <Arduino.h>#include <DYPlayerArduino.h>#include <SoftwareSerial.h>
// Initialise on software serial port.SoftwareSerial SoftSerial(10, 11); // RX, TXDY::Player player(&SoftSerial);
void setup() {  player.begin();  // Also initiate the hardware serial port so we can use it for debug printing  // to the console..  Serial.begin(9600);
  Serial.print("Initializing: ");
  player.setVolume(30); // 50% Volume  player.setCycleMode(DY::PlayMode::REPEAT); // Play all and repeat.  player.play();}
void loop() {
  // Print the number of the sound that is playing.  Serial.print("Playing sound: ");  Serial.println((int16_t)player.getPlayingSound());  delay(500);}

Output displaying in the Serial Monitor while music is playing:
Initializing: Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0

_________________________________________________________________

All we have to decide is what to do with the time that is given to us.

daPoppa, 2020-04-01:
Here are some snaps of the setup.

_________________________________________________________________

All we have to decide is what to do with the time that is given to us.
 

    On Wednesday, April 1, 2020, 03:15:46 p.m. PDT, Harold Johnson <ha_johnson@yahoo.ca> wrote:  
 
 Thnx for the hints. 
Yes, I had the usb cable connected, which I thought made sense, since I was using different pins 10/11 and the scripts says to use the monitor window for debugging.  How could I use debugging if the usb cable is disconnected?  
I'm using an Aduino Uno (knockoff), connected to pins 10/11, using the same script that you provided (thank you).  I have 1k resistors between both TX/RX - using the USB to power the arduino atm.  
Sorry, no access to an oscilloscope. 
Yes, this board appears to allow PWM~ on pins 10/11.  I have used those pins before with neopixels. 

#include <Arduino.h>#include <DYPlayerArduino.h>#include <SoftwareSerial.h>
// Initialise on software serial port.SoftwareSerial SoftSerial(10, 11); // RX, TXDY::Player player(&SoftSerial);
void setup() {  player.begin();  // Also initiate the hardware serial port so we can use it for debug printing  // to the console..  Serial.begin(9600);
  Serial.print("Initializing: ");
  player.setVolume(30); // 50% Volume  player.setCycleMode(DY::PlayMode::REPEAT); // Play all and repeat.  player.play();}
void loop() {
  // Print the number of the sound that is playing.  Serial.print("Playing sound: ");  Serial.println((int16_t)player.getPlayingSound());  delay(500);}

Output displaying in the Serial Monitor while music is playing:
Initializing: Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0Playing sound: 0

_________________________________________________________________

All we have to decide is what to do with the time that is given to us.

miamiusa1234, 2025-08-21:
Hi SnijDerC, any chance you can share the sketch you used to test it, thank you