SnijderC/dyplayer #13: Arduino Nano 33 IOT issue

jvet88, 2020-03-13:
Hi, I'm trying to use the library with a Arduino Nano 33 IOT and get the following error when compiling:
D:\Arduino\libraries\DYPlayerArduino/DYPlayer.h:392:1: error: expected declaration before '}' token
 }
 ^
exit status 1
Error compiling for board Arduino NANO 33 IoT.

Compiling for Uno works properly, do you have an idea on what could be causing the problem?

SnijderC, 2020-03-28:
Can you send me your code? I don't see any problem with the code this error applies to. Are you using the exact same code for for both boards or are you maybe using some defnitions to detect different boards during compilation?

jvet88, 2020-03-31:
I was using your sample code, maybe you can try if you select the Arduino NANO 33 IoT board and try to compile. 
I believe it has something to do with the serial connection, as I was able to copy the relevant functions to a new project and got it to send the desired commands to the DYplayer

SnijderC, 2020-03-31:
It appears the problem is way up in the trace, the problem is that the headers for that board define a macro called `USB` and this library has an enum that defines `USB` as a device type (of storage devices on modules, `FLASH`, `USB`, `SD`, etc. The preprocessor runs the macro which I guess inserts a bunch of code to initialise a USB device or something. Which means the definition of the `device_t` type fails, then we have dangling braces which causes the error you had. It's not easy to fix, I either have to rename the `USB` device type or find a way to prevent the macro from running when the device type is created (I'm not aware of that possibility). Also if I do the latter, any time a developer uses the `DY::Device::USB` type, everything comes tumbling down again.. Will think about  what to do and investigate a bit more.

SnijderC, 2020-04-01:
I could succesfully compile the version of this library in this branch: https://github.com/SnijderC/dyplayer/tree/ft-prevent-naming-collisions

Can you check that this works for you?

You will have to change any occurance of: `DY::PlayState::REPEAT`, `DY::Device::FLASH`, etc. by `DY::PlayState::Repeat`, `DY::Device::Flash` (note the case change). If this works for you I will merge this and everybody that uses the new version will have to do the same.. So it's invasive but definitely an improvement.

closed 2020-07-08 by SnijderC