SnijderC/dyplayer #14: Compiling via PlatformIO fails

ghoeffner, 2020-03-22:
Trying the example sketch at https://github.com/SnijderC/dyplayer/blob/master/examples/PlayAllSounds/PlayAllSounds.ino compilation on platformio for nodemcuv2 (esp8266) fails in a weird way. Am I doing something the wrong way?
"DYPlayer @ 3.0.0 has been successfully installed!" -> seems like it should be installed...
Would appreciate any help :)

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.3.3 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Looking for DYPlayer library in registry
Found: https://platformio.org/lib/show/6949/DYPlayer
LibraryManager: Installing id=6949 @ ~3
Using cache: C:\Users\German\.platformio\.cache\64\0f25ce840531ee206a1ffe745b719c64
DYPlayer @ 3.0.0 has been successfully installed!
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DYPlayer> 3.0.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\nodemcuv2\lib788\DYPlayer_ID6949\dyplayer\DYPlayer.cpp.o
Compiling .pio\build\nodemcuv2\lib788\DYPlayer_ID6949\dyplayer\DYPlayerArduino.cpp.o
Compiling .pio\build\nodemcuv2\lib788\DYPlayer_ID6949\dyplayer\DYPlayerESP32.cpp.o
Archiving .pio\build\nodemcuv2\libFrameworkArduinoVariant.a
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp-frag.cpp.o
src\main.cpp:2:29: fatal error: DYPlayerArduino.h: No such file or directory

*************************************************************************
* Looking for DYPlayerArduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:DYPlayerArduino.h"
* Web  > https://platformio.org/lib/search?query=header:DYPlayerArduino.h
*
*************************************************************************

 #include "DYPlayerArduino.h"
                             ^
compilation terminated.
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp-version.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FS.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FSnoop.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FunctionalInterrupt.cpp.o
*** [.pio\build\nodemcuv2\src\main.cpp.o] Error 1
===================================================================================== [FAILED] Took 3.31 seconds =====================================================================================
The terminal process terminated with exit code: 1
```

SnijderC, 2020-03-28:
I see why this happens, I'll move the source from `src/dyplayer` to `src` which both Arduino IDE and platformIO expect, then it will be fixed. I did it for testing but I'll have to figure out how to do that differently. For now you can work around this by including `dyplayer/DYPlayerArduino.h` instead. If you install a later version you will have to change that back though.

SnijderC, 2020-03-29:
Master should now not have this issue any more I will make a new release to platformIO soon.

ghoeffner, 2020-03-29:
Awesome, thanks for your help @SnijderC 🎉

closed 2020-03-30 by ghoeffner