SnijderC/dyplayer #16: Update readme on odd path requirements and hidden files

SnijderC, 2020-03-28:
Paths for combination play seem to have to be required to be named `ZH/`, as opposed to `DY/` as in the manual of the modules.

Files played by number are not corresponding to the number in the file name that, while it is suggested in the manual to number the files. It seems to be that the first file found on the drive is what is played when you ask it to play `00001`, even if that is a file in a `.Trash` directory.

ghoeffner, 2020-03-31:
Files named x... are FIRST not LAST like you would expect! 
[image]

SnijderC, 2020-03-31:
Did you put that file first? Or do you have the same file in a hidden trash directory, and was it the first you saved on the module? In my experience it plays the files in order of appearance on the disk.

ghoeffner, 2020-03-31:
It is definitely a "bug" (or intended behavior...) of the board. All files are saved to an SD card, my module does not have embedded memory. I'd probably add to the readme, that having additional files (whatever name they have...) is not recommended, because playing by ID relies on the file order and the sorting algorithm used is unknown?

SnijderC, 2020-04-01:
I added some lines on it in the README, do you think it makes sense?

closed 2020-07-08 by SnijderC