SnijderC/dyplayer #17: Add tested boards to the list in README

SnijderC, 2020-03-30:
You all recently made issues on this project, and I've tried to help you out with them. Could you do me a favour and let me know which boards you used and if they worked okay? There is a list in the [readme](https://github.com/SnijderC/dyplayer/blob/master/README.MD) that I would like to update with your experiences so others can find out if their boards should work with this library.

If you could also tell me what boards you were using, i.e. Arduino boards or other (and which type) that would be super helpful!

Thanks a lot for your feedback!

@ntapsis @daPoppa @jvet88 @ghoeffner

ghoeffner, 2020-03-31:
Hey @SnijderC,

DY-SV5W is tested and working. I am wondering if this is a bug or not, but specifying   player.playSpecified(9);
is playing song 00007.mp3 - is this intended behavior or a bug? Or some board incompatibility?

SnijderC, 2020-03-31:
I see you found issue #16, that is exactly about this issue. So let's discuss it there. It is not a bug in this library I think, I think it's a "bug" or odd behaviour of the module. Thanks for sending the module name. Can you maybe share the board you are connecting the module to as well?

ghoeffner, 2020-03-31:
DY-SV5W connected to a nodemcu v2, tested using swserial on D3 and D4 and using hwserial on Serial1. Additionally, here are pictures - feel free to use them for the readme if you want.

[image]
[image]

SnijderC, 2020-04-01:
Nice. thanks! Perhaps I'll add a gallery at some point but for now we have only 2 images ;)

Doubletop12, 2020-07-07:
DY-SV5W with PIC 18F27K42. Code modified to be used with Microchip XC8 compilier it works fine.

See my post about the 1k resistors in the Tx/Rx lines.

SnijderC, 2020-07-08:
> DY-SV5W with PIC 18F27K42. Code modified to be used with Microchip XC8 compilier it works fine.
> 
> See my post about the 1k resistors in the Tx/Rx lines.

Maybe you can add a wiki page on how to do that?

Doubletop12, 2020-07-08:
Chris

I'll have a look at what I did, it wasn't that hard but my C isn't industrial  quality and I don't do C++. I just removed the class identifiers, made some fairly minor work arounds when the complier complained. 

I'll see if I can do a diff of my source and yours and see if I can summarise the changes.

Pete

woutput, 2021-06-23:
[image]
I created this toy for children using [Wemos D1 mini clone](https://www.aliexpress.com/item/32659152063.html) and [DY-SV17F](https://www.aliexpress.com/item/4000130060681.html).
Note that I applied the changes in #37

closed 2022-10-27 by SnijderC