SnijderC/dyplayer #18: 5V to 3.3V Tx/Rx lines

Doubletop12, 2020-07-06:
I'm trying to confirm the need for the 1K resistors in the Tx/Rx lines with the DY-SVxxx cards or whether those devices have them installed.  I'm using dyplayer, on a PIC with, the code converted to C and compilied with XC8 in Microchips MPLABX IDE. It all compiles well and test code sends the correct byte sequences to the PIC Uart. All good

The PIC (and I assume Arduino) use 5v signal levels.  The data sheets for the similar XY-V17B advise the use of the 1k resistors in the lines, those cards don't have any components in the signal lines but the DY-SVxxx modules appear to have conditioning resistors already installed. When I did put a 1k resistor in the PIC Tx to DY_SV5W Rx line the signal  at the DY-SV5W end was severly distorted.  

In addition if a 1K resistor is put in the PIC/Arduino Rx line wouldn't that reduce the signal level when the PIC/Arduino is expecting to see  a 5V Rx signal?

Any clues anybody? I can't get this to work and may have lost one of my DY-SV5W cards in trying to find out?

Doubletop12, 2020-07-07:
A bit of an update. I thought I'd try again with my dead DY-SV5W card, today it was working fine. It may have been a power supply problem. I had been powering by the USB connector and I'm sure I've done that in the past but it doesn't seem to work despite being able to read the files on the uSD card.

Today I used the 5v-0v pin for power and the dyplayer Tx works fine. However, it doesn't work with a 1k resistor in the line. 

I've not tried Rx yet

SnijderC, 2020-07-08:
[image]

Those SMD resistors marked `01B` are in fact 1K resistors, I am not sure which pins they connect to but their placement suggests that they are indeed connected to the `TXD/IO0` and `RXD/IO1` pins maybe you can check that? 

It's odd because all the IO pins can be used to connect buttons to, to each play 1 sound on the module, you would expect them to all have the same configuration. They do all seem to have a small capacitor connected too, maybe to debounce the buttons? I can imagine 2 resistors and a capacitor making an RC filter and causing your UART to form a kind of sawtooth wave shaped signal.

You could also maybe follow the path to the pin on the chip and measure the resistance there. Or you could measure the voltage there but you would need to use an oscilloscope to check the amplitude, because a voltmeter will only measure an average voltage, which will always be lower than the signal voltage.

> In addition if a 1K resistor is put in the PIC/Arduino Rx line wouldn't that reduce the signal level when the PIC/Arduino is expecting to see a 5V Rx signal?

It depends, AVR needs `0.6 * Vcc` to consider a signal a logic level `1`, I don't know what your PIC needs, refer to the datasheet.
It will have some hysteresis that makes sure anything between almost 0V and 0.6Vcc is considered in transition (the logic level will not change within that range) and anything below that level is considered to be logic level `0`.

In the case of AVR, `0.6*Vcc = 0.6 * 5 = 3.0V`, 3.3V is enough above that to be okay most of the time. However that assumes your 5V  power supply is not 5.5V and your 3.3V is not 3.1V, especially if they are both off by a bit it can be on the limit.. If you're not sure, use a proper logic level converter.

Doubletop12, 2020-07-08:
Chris

I've been down that route and had discovered the on board 1k resistors but I can't find a datasheet for the DY-SV5W.  I have a pinout for a XY-V17B but although it looks to have similar logical functions, I can't assume the pins are the same. I'll take another look but the assumption I am making is the Tx/Rx lines have 1k resistors fitted. If logic levels are being used to drive the other IO pins they will need some resistors.

I agree that the resistor capacitor combination may be unique to the Tx/Rx lines and that may be causing the signal distortion if another 1k resistor is included in the line. I have seen a diagram with capacitors to VSS on all the I/O lines. 

At a quick glance a PIC need 2.0V to be seen as a 1. With the DY-SV5W Tx being 3.3V and most likely going via a 1k resistor a in line resistor going to the host would seem unnecessary.

SnijderC, 2020-07-08:
> At a quick glance a PIC need 2.0V to be seen as a 1. With the DY-SV5W Tx being 3.3V and most likely going via a 1k resistor a in line resistor going to the host would seem unnecessary.

Oh, no you don't need one on the TX of the module (RX of the PIC) for sure.

SnijderC, 2020-07-08:
* [ ]  make a reference to this issue in the readme.

SnijderC, 2020-07-08:
@Doubletop12 Looks like this in this setup the resistors are also not present: https://github.com/SnijderC/dyplayer/issues/17#issuecomment-606885470

Doubletop12, 2020-07-09:
@SnijderC Looks like this in this setup the resistors are also not present: #17 (comment)

You mean the additional inline resistors aren't there. They are there on the board for the Tx/Rx lines but not the other IO lines. 
If using IO2-7 inline resistors are required 

It really depends on the style of board you are using. Some of the other styles have them on TX/Rx but not on the IO2-7. We really need a datasheet for the DY-SVxx to be able to get a definitive answer but it appears that users need to confirm the type of card they are using and confirm for themselves that the resistors are present.

These cards appear to have them
https://ae01.alicdn.com/kf/HTB1u0den_Zmx1VjSZFGq6yx2XXad/DY-SV17F-MP3-Io-USB.jpg_q50.jpg

SnijderC, 2022-10-27:
It's been a long time, but I've added some lines to the README.

closed 2022-10-27 by SnijderC