SnijderC/dyplayer #20: ONE Line protocol support.

Vallerik, 2020-08-16:
Hello! Great library, I am delighted but also frustrated at the same time. Since there is no support for the ONE Line protocol. Can you add it? I really wanted to deal with him and control the player using only one wire.

SnijderC, 2020-08-18:
Hi Vallerik, 

Unfortunately that protocol is entirely different from the UART protocol, not only in its hardware interface, but also every command and parameter is different. Instead I suggest you to just connect a UART TX pin from your microcontroller to the RX pin on the sound module, you will not get any feedback, so don't use functions that return a value from the module. But you have the same level of control with as many wires as the ONE Line protocol. 

Does that help?

Vallerik, 2020-08-19:
Hello!  Thanks for the answer!  I understood you.  But I want to deal with the ONE Line protocol, for some of my Arduino projects this connection is needed.  Could you help me?  Will make at least a simple example for sending one command over this protocol?

С уважением,
Валера Залит
valera.zalit@inbox.ru

SnijderC, 2020-08-19:
I'm not adding support to this library, also not through pull request because the protocols are too different.

There is a manual floating around the internet with instructions on each protocol, hardware connections etc. If you don't have that manual, I can maybe help you find that, does that help you?

Vallerik, 2020-08-19:
Good. I realized that you will not add this protocol.
I just wanted to ask for a simple example of a one line control program. If it's not difficult for you, please send a manual or a simple example program.

> I'm not adding support to this library, also not through pull request because the protocols are too different.
> There is a manual floating around the internet with instructions on each protocol, [...]
 
 
С уважением,
Валера Залит
valera.zalit@inbox.ru

closed 2020-09-08 by SnijderC

arduino12, 2022-05-24:
**Hi,**

@SnijderC  I just made a simple library to control the DY_SV17F module that I got [**dy_sv17f_wire**](https://github.com/arduino12/dy_sv17f_wire),

It will be helpfull if you can drop a link to my repo in this repo README,
Anyway I have added a link to your repo in mine :)