SnijderC/dyplayer #23: Getting back file names from USB disk

TheSap, 2020-11-16:
Hi SnijderC, thanks for the great work done! I'm a rookie in that and I got lost somewhere and couldn't understand which board can give back list of files present in the U Disk and not only the count of files.
Reading at what you wrote I understood that this is possible but, then, I couldn't find any more mention of this capability.
Thank again for your time!

SnijderC, 2020-11-17:
Hi,

The board doesn't use file names at all except for the option to play a sound by name. It will not list sound files, there is only a function that will [get you the sound number](https://github.com/SnijderC/dyplayer/#uint16_t-dydyplayergetplayingsound). So you will have to keep a list of file names in your source code and do a lookup. 

Hope it helps.

TheSap, 2020-11-18:
Many thanks for the answer. Unfortunately it is not what I was looking for. I'm building an audio device, using Arduino to control all the functions, with Fm radio, arduino controlled pre-amplifier and input selection and I'm still missing the USB mp3 play with a "real" control like the cheap device you may find around showing names... I will keep serching. Thanks again

closed 2021-04-27 by SnijderC