SnijderC/dyplayer #24: Commands working fine however can't get valid response

Orel-A, 2020-12-28:
I'm using ESP8266 connected to DY-SV17F. I'm able to send commands to the module but attempting to use the functions checkPlayState / getPlayingSound I'm unable to get a valid response. Seems like I get garbled data. Not sure how to further debug this as I don't have any ideas as to why sending data is working fine but getting back data not.

SnijderC, 2020-12-28:
Can you tell me how you connected the module? Did you use a resistor on the RX line?

Orel-A, 2020-12-29:
I followed the instructions here in the segment "Wiring the module". I didn't use a resistor on the RX line.

Orel-A, 2020-12-29:
Well, right now it's not too inportant as I've managed to solve my problem by using the Busy pin on the module.

closed 2020-12-29 by Orel-A