SnijderC/dyplayer #28: Only odd files are playing

sorokonoz, 2021-03-20:
Hello
I'm completely lost here. Prepared SD card according to instructions. Formatted to FAT, copied ten files one by one named 00001.mp3 etc. to the root folder. Using this function "player.playSpecified(8);"

The result is puzzling - only odd files are playing. This strange pattern is making me curious. 
00001.mp3 = ok
00002.mp3 = not
00003.mp3 = ok
00004.mp3 = not
00005.mp3 = ok
00006.mp3 = not
00007mp3 = ok
00008.mp3 = not
00009.mp3 = ok
00010.mp3 = not

Changed the CD card. No change. Files are ok, playing on mac.
Formatted card and copied the back in different order.
Pattern stays the same.

Appreciate any ideas!

Janis

solhuebner, 2021-03-20:
Have you checked https://github.com/SnijderC/dyplayer#important-files-do-not-play-in-order-of-named-sequence as it might be better to use another function?

sorokonoz, 2021-03-21:
I suspect that I have device with different or corrupted firmware. Path play doesn’t work at all, even when I do the cooy/paste from examples folder. But instead, the playSpecified is playing files not by sequence they were copied to the drive, but by numbers in the names. Just odd numbers. So I made workaround by utilising this pattern, renamed files to have only odd numbers in names and now I can select and play them all. Very strange :) I don’t think that the bug is in your library, something is wrong with the device. Thanks for replying! :)

Janis

closed 2021-03-21 by sorokonoz