SnijderC/dyplayer #29: How are the mp3 files loaded into the flash memory of the DY-SV17F?

maxSnyder, 2021-03-30:
Hi Chris, 
first off, let me tell you that you really did an excellent job. I also ordered a DY-SV17F board, but it is still in the mail and has not yet arrived. But I am already thinking about the handling and I have one question: I am not sure how I can load my mp3 files into the flash memory on the DY-SV17F. I understood your explanations to mean that I simply connect the board to my PC using a USB cable and a new CD-ROM drive then appears in the PC's file explorer. Is that so and can I then simply copy my mp3 files and paste them into this CD-ROM directory? Will the IC DY1703A on the DY-SV17F board take over the entire flash programming process or do I need any special software? 
And what is the reason for the file "UART tuning tool.exe", that I found elsewhere on the Internet? I have opened the exe file, a window appears with a lot of Chinese characters that mean little to me.
With best regards
maxSnyder

woutput, 2021-04-15:
Hi @maxSnyder Yes, you can just connect your PC via USB to the connector of the module; you'll see a 4 MiB flash drive (just like a USB stick) where you can put your MP3 files. You have to rename them to 00001.mp3 00002.mp3 etc. Just (safely) disconnect after copying the MP3 file(s) and start playing around. No need to use "UART tuning tool.exe"

closed 2021-04-27 by SnijderC

mohsencherfl, 2023-01-31:
Hi @SnijderC,
I was testing this module to see how it works, however when I did connect it to the computer through a usb port it was not recognized by the operating system, then I realized that the board was getting hotter and hotter so pulled the usb cable out. When I tried it out for the second time it did nothing as if it was dead. Is it possible that my module has burnt out?