SnijderC/dyplayer #30: ESP32 Wrover module issue

domconscious, 2021-04-07:
I am attempting to use the "DYPlayerESP32.h" module on an ESP32 Wrover but get a compile issue below.

It seems to choke on this line:
DY::Player player(UART_NUM_2, 16, 17);

Any help would be off assistance!

```
sketch/BT_RC_Trim.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_player+0x4): undefined reference to `DY::Player::Player(uart_port_t, unsigned char, unsigned char)'
sketch/BT_RC_Trim.ino.cpp.o: In function `_GLOBAL__sub_I_player':
Multiple libraries were found for "WiFi.h"
 Used: /Users/xxx/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
/Users/xxx/Documents/Arduino/Fartrarri_V1/BT_RC_Trim/BT_RC_Trim.ino:13: undefined reference to `DY::Player::Player(uart_port_t, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Wrover Module.
```

solhuebner, 2021-04-08:
Can you post the snippet where you initialize your Player object?

domconscious, 2021-04-08:
```
// Initialise the player on uart2, pins TX: 16, RX: 17.
DY::Player player(UART_NUM_2, 16, 17);

```
Unless I am missing something else under setup?  I am basing this off of the examples/esp32/PlaySounds.cpp

domconscious, 2021-04-10:
I believe I've figured this out on my own.  This example was not meant for Arduino but only for ESP-IDF.

For the ESP32 specifically I am trying to sort out how to use an alternative serial port in order to retain the standardised serial output for troubleshooting.  Any help there would be awesome.

SnijderC, 2021-04-27:
Looks like you intend to use it with Arduino, then look at the [SoftwareSerial example](https://github.com/SnijderC/dyplayer/blob/master/examples/SoftwareSerial/SoftwareSerial.ino). Then you can use pins that are not the hardware Serial port and thus you can use the hardware serial port for debugging.

closed 2022-10-27 by SnijderC