SnijderC/dyplayer #34: 'HardwareSerial*' to 'DY::Player&&' Conversion error (platform.io and arduino ide)

mo-norant, 2021-06-01:
setup:
platform io 
esp32

```
...
DY::Player player(&Serial2);
...
```
i am using pin 16 and 17 on the board. So i can't use the regular DY::Player player.
when sending a basic play command by UART the module does respond. so it has to do with the configuration of the serial interface in the constructor i think

when compiling the "PlaySoundByNumber.ino" to an esp32, the following error is thrown:

```
lib\dyplayer-master\src/DYPlayerArduino.h:15:7: note: candidate: DY::Player::Player()
       Player();
       ^
lib\dyplayer-master\src/DYPlayerArduino.h:15:7: note:   candidate expects 0 arguments, 1 provided
lib\dyplayer-master\src/DYPlayerArduino.h:11:9: note: candidate: constexpr DY::Player::Player(const DY::Player&)
   class Player: public DYPlayer {
         ^
lib\dyplayer-master\src/DYPlayerArduino.h:11:9: note:   no known conversion for argument 1 from 'HardwareSerial*' to 'const DY::Player&'
lib\dyplayer-master\src/DYPlayerArduino.h:11:9: note: candidate: constexpr DY::Player::Player(DY::Player&&)
lib\dyplayer-master\src/DYPlayerArduino.h:11:9: note:   no known conversion for argument 1 from 'HardwareSerial*' to 'DY::Player&&'
```

@IrregularShed @SnijderC @chatelao

agfaps, 2022-01-01:
Hi @mo-norant ,
Have you found solution/workaround for this?
It's my first time using dyplayer with esp32 arduino and i'm also facing the same problem.
I intend to use hardware serial port 2 to communicate with dyplayer.
Thanks.

agfaps, 2022-01-01:
Hi @mo-norant ,
I have managed to play mp3 file in /01/0001.mp3 using Serial2 in ESP32 in Arduino IDE.
I just change Serial to Serial2 in DYPlayerArduino.cpp:
`namespace DY {`
`  Player::Player() {`
`    this->port = &Serial2;`
`    this->isSoftSerial = false;`
`  }`
and in Setup:
`  // Serial2 for Dyplayer`
`  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);`
`  while (!Serial2);`
`  player.begin();`
`  player.setVolume(15); // 50% Volume`

SnijderC, 2022-10-27:
@mo-norant Sorry it's been so long you had no reply, I'm going to assume you're not still waiting for an answer, but if you do still want help, feel free to reopen this.

closed 2022-10-27 by SnijderC