SnijderC/dyplayer #36: DY::checkPlayState

ozhally, 2021-06-12:
Hello.
Can I have an example on how to use the player state function in Arduino on the DY-SV5W please?

DY::Player player;

void loop() {
if (player.checkPlayState.Playing()
    player.stop();
}

Thanks

srauco, 2022-01-20:
I'm also having an issue with this. I've written the code so it doesn't throw an error, however it does not return the correct play state

SnijderC, 2022-10-28:
This should blink 2 times when playing, 3 times when paused and 1 time when stopped, works just fine here.
```
#include <Arduino.h>
#include "DYPlayerArduino.h"

// Alternatively initialise on another serial port.
DY::Player player();
void setup()
{
  player.begin();
  player.setVolume(15); // 50% Volume
  player.setCycleMode(DY::PlayMode::Repeat); // Play all and repeat.
  player.play();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{

  for (int i = 0; i < ((int)player.checkPlayState()) + 1; i++)
  {
    delay(500);
    digitalWrite(LED_BUILTIN, 0);
    delay(500);
    digitalWrite(LED_BUILTIN, 1);
  }
  delay(5000);
  player.pause();
  for (int i = 0; i < ((int)player.checkPlayState()) + 1; i++)
  {
    delay(500);
    digitalWrite(LED_BUILTIN, 0);
    delay(500);
    digitalWrite(LED_BUILTIN, 1);
  }
  delay(5000);
  player.stop();
  for (int i = 0; i < ((int)player.checkPlayState()) + 1; i++)
  {
    delay(500);
    digitalWrite(LED_BUILTIN, 0);
    delay(500);
    digitalWrite(LED_BUILTIN, 1);
  }
  player.play();
  delay(5000);
  for (int i = 0; i < ((int)player.checkPlayState()) + 1; i++)
  {
    delay(500);
    digitalWrite(LED_BUILTIN, 0);
    delay(500);
    digitalWrite(LED_BUILTIN, 1);
  }
  delay(5000);
}
```

closed 2022-10-28 by SnijderC