SnijderC/dyplayer #38: Delay between player.begin() and player.setVolume required

woutput, 2021-06-23:
Hi,

I have these

```
#include <SoftwareSerial.h>

SoftwareSerial SoftSerial(D1, D2); // RXpin, TXpin
DY::Player player(&SoftSerial);
```

and my `void setup()` ends with:

```
    player.begin();
    delay(100);
    player.setVolume(15);
    player.setCycleMode(DY::PlayMode::OneOff);
    player.stop();
    Serial.println("Setup done");
```
but if I remove the `delay(100)` after `player.begin()` then the setVolume command has no effect.
I'm not sure if this is an issue of this library since I've seen similar behavior after `Serial.begin()` but I think that it should still be solved. Maybe it can be solved by adding the `delay(100)` inside the `player.begin()` function?
In addition: maybe other calls to player.xxx directly after `player.begin()` also don't work; I didn't test that. I also didn't try values lower than 100 ms.

I'm using ESP8266 on Wemos D1 mini clone and DY-SV17F

Nightreaver, 2021-12-24:
I can confirm this, was struggling to set the volume, but after adding the delay, it worked well.

SnijderC, 2022-10-27:
Seems to work fine after adding a check to see if Serial is ready.

closed 2022-10-27 by SnijderC

woutput, 2022-10-28:
> Seems to work fine after adding a check to see if Serial is ready.

Nice!
Should this check be included inside player.begin() ?
Or at least added to the docs?
Thanks

SnijderC, 2022-11-01:
No, no need to add any lines, there is a loop in the begin call now that waits for the Serial connection to establish and then continues.

Gorkde, 2025-04-09:
I just updated my Arduino IDE as well as the boards.
Just after that my formerly perfectly working script stopped working correctly and was now always playing in full volume.

Because I found this old issue I did also try to put a delay in between
    
```
player.begin();
delay(300);
player.setVolume(15);
```

And suddenly it started working correctly again and I could set the volume again now.

So there must be still an issue here! And it has nothing to do with the serial, seems you guys misunderstood each other.