SnijderC/dyplayer #42: playSpecifiedDevicePath()

cstme, 2021-11-12:
Thanks for your awesome work.
 I'm using a SV5W connected to an "arduino" pro micro. 
I can make most commands work except for playing by path. I've got three files in /root 108.mp3, 109.mp3 and 110.mp3 and 3 directories with numbered files - /101 - /102 - /103.  I've tried for a week to get the play by path to work but with no success.  Can't even select the root files let alone the ones in folders.  Is any one else able to make this work.   
Just want to confirm the command is working for someone before spending any more time on it.
Any feedback or tips would be much appreciated. I'm out of ideas.

solhuebner, 2021-11-12:
@cstme you can have a look at #35 as it was fixed but there was no new release so you might need to download the updated file and replace it

cstme, 2021-11-13:
Thanks for that.  It seems my file already has the update but I'll take a closer look over the weekend.

solhuebner, 2021-11-24:
@cstme do you still have the same issue?

closed 2022-10-27 by SnijderC

tr00p3r, 2024-12-29:
Sorry to necro but this was bugging me.

What works for me is removing the "." in ".mp3"

String path = "/MYFILE" + String(index) + "MP3";
player.playSpecifiedDevicePath(DY::Device::Sd,path.c_str());