SnijderC/dyplayer #43: SoftwareSerial only for AVR?

jenschr, 2021-12-29:
Hi,
I wanted to use this library with an ESP32 S2 module with the Arduino IDE and I noticed that SoftwareSerial is only available on AVR? Any specific reason for this, given that SoftwareSerial also exists on many other platforms? (Here's the ESP32 one https://www.arduino.cc/reference/en/libraries/espsoftwareserial/ and it works identically to the AVR version)

Removing all the #ifdef __AVR__ will solve this. I'm also a bit sceptical to the use of #ifdef HAVE_HWSERIAL0. On my ESP32 S2 dev board from Unexpected Maker, this is not defined despite the board having hardware Serial1 and Serial2 (ref https://www.youtube.com/watch?v=3sXtVuMhuoc). Removing all the ifdef's actually made this library work much better.

SnijderC, 2022-10-27:
For some boards it will actually not work, because for example they don't define `SoftwareSerial.h` at all. So this cleverness is actually necessary. That said, I've updated the cleverness, it should be more reliable now.

closed 2022-10-27 by SnijderC

jenschr, 2022-11-04:
+1 for more cleverness!