SnijderC/dyplayer #44: PlaySpecified, getPlayingSound numbers do not match file numbers

ntewinkel, 2022-01-03:
Using DY-SV17F soundboard with Arduino Uno.
On the soundboard I've placed files 00001.mp3, 00002.mp3, 00003.mp3 ... up to 00009.mp3

When playing all the sounds using the SoftwareSerial example (included in this library), it displays the sound numbers being played as 1, 3, 5, 7, 11, 13, 15, 17

Similarly for PlaySpecified, I have to use those odd-number representations instead of the actual number.

My workaround for PlaySpecified is to convert the requested sound number:
  int convertedSoundNumber = (soundNumber-1) * 2 + 1;
  player.playSpecified(convertedSoundNumber);

srauco, 2022-01-20:
Playing by sound number actually plays the sounds in the order they were loaded on the board, not file number. Play by file name instead

SnijderC, 2022-10-28:
Checked this again, I think @srauco is right, seems to work fine, but oddly I will get back a different number than I sent it to play when I call `getPlayingSound`. Fairly sure this is due to the order of the files on disk. Since the board doesn't care where a file is, only in which order it was written to the disk, check that you don't have a hidden trash directory on the disk.

closed 2022-10-28 by SnijderC