SnijderC/dyplayer #45: DY-SV8F fonctional with serial com

sbarabe, 2022-03-24:
Hi !

If it can be of any help, I'm working on a project using the DY-SV8F with an arduino Nano and so far the module is responding good to the library. I'm not an expert so I can't say that it's completely tested, but it's functional.

The only thing I've encountered that is bugging me is that I've try to use the Nano Tx/Rx pins but it didn't work. I'm using two other pins instead (D9/D10) and it works great. As I have said, maybe it's just something I'm doing wrong since I'm an hard working amateur...

Also, I'm using the audio jack to send the signal to a small amp the to two speakers. Works great and you can adjust the sound volume by software with the library.

Thanks for the great work !
[image]

Tezzatron81, 2022-03-24:
You are putting tx to rx on the board. And rx to tx. I have dysv5w and it works with the nano

sbarabe, 2022-03-24:
Ok  so I plug the DY Tx pin to the Nano Rx, and the other 2 togheter? Thinking of it it makes great sens... I should have seen it this way !

Tezzatron81, 2022-03-24:
yes. works on minw and pretty sure most these boards are the same

sbarabe, 2022-03-24:
So just to be sure, when you define the softserial(serial1&2), witch pin is the first to be define, the Tx or Rx of the Nano ?

Tezzatron81, 2022-03-25:
I dont define them at all, i just use the modified version of the play specfied example. It works for me

sbarabe, 2022-03-25:
Ok works great with only the " DY::Player player; " definition with Nano Rx/Tx pins connected correctly to the DY-SV8F Rx/Tx pins.
Thanks for the help !

As said here on the readme file, if all files are played, the files are playing in the order I have copied them on the DY board, not by files name...

closed 2022-04-01 by sbarabe