SnijderC/dyplayer #47: PlayRandom always starts from the first song in the folder

vulterey, 2022-05-25:
Hi,

I am using Arduino Uno R3 with DY-SV5W Voice Playback Module and it seems that PlayRandom option doesn't work as I think it should with this setup.
I've tried to use my own sketch and also just PlayRandom.ino alone to check if this is my mistake or some bug, but no difference.
I've checked the connection between Tx/Rx ports on the boards, but it looks ok and other functions like playSpecified work perfect but not the Random option. The files are named 00001.mp3 to 00010.mp3 and placed on the root SD card. The module is powered by Arduino.

Whenever I power on the Arduino the DY-SV5W module plays the first song then a random track is picked after.
But when I reset Arduino after the DY-SV5W module moves over to the next song (for example 2), then the module starts from this (2) song. Then again the module picks randomly the next song.

I will probably overcome it with the generation of a random number from the 1-10 range and use playSpecified(random_number) but I am curious if this is the right way for PlayRandom function as I didn't find any info in the resources files to explain it.

Any ideas?

Regards
Adam

SnijderC, 2022-10-27:
Most of microcontroller don't really have the capacity to come up with truly random numbers, they don't have true random generators, and to use pseudo random generators you need a seed, some random data to get started with. You'll see computers use all sorts of sources, network traffic for example. But boards like this don't have any sources, so I fully expect these to just play the same "random" sequence every time you restart it.

closed 2022-10-27 by SnijderC

vulterey, 2022-10-28:
Hi, Thank you for the answer. 
I used randomSeed(analogRead(A4)); to do a pseudorandom number generation from analogue port.
It will be initialized from the “noise” on the port when nothing is connected.

SnijderC, 2022-11-01:
That's how its done!