SnijderC/dyplayer #48: Play specific files according to their actual filename.

Bungalore, 2022-05-31:
Hi. 
I don't know if it is the place to post this, but as the player plays the files in the order they are written in partition (i.e. file 1 for the first file in the partition and not for the actual 00001.mp3), it is actually possible to match each file number with it's actual filename, using a freeware called [Drivesort](https://www.anerty.net/software/file/DriveSort/?lang=en), which sorts files in the partition without the need of formatting and copying them in the right order.

I use this software to sort USB flash drive for my car built-in mp3 player.

I hope this helps

SnijderC, 2022-10-27:
Thanks for the tip!

closed 2022-10-27 by SnijderC

JakesMD, 2025-03-20:
Hi! The same exists for Mac: [FatDriveSorter](https://fat-drive-sorter.netlify.app). Works a charm. I suggest you add this to the README. I was lucky to find this issue.