SnijderC/dyplayer #49: SoftwareSerial does not name a type

PaulMPapa, 2022-11-22:
Hi there,

Thanks for your effort and sharing of the library. 

Although i managed to connect and play sound, I am trying to use the software serial, but i can not get passed the compilation. The error message states :

SoftwareSerial:6:1: error: 'SoftwareSerial' does not name a type; did you mean 'HardwareSerial'?
 SoftwareSerial SoftSerial(10, 11);
 ^~~~~~~~~~~~~~
 HardwareSerial
SoftwareSerial:7:20: error: 'SoftSerial' was not declared in this scope
 DY::Player player(&SoftSerial);
                    ^~~~~~~~~~
C:\temp\SoftwareSerial.ino:7:20: note: suggested alternative: 'Serial'
 DY::Player player(&SoftSerial);
                    ^~~~~~~~~~
                    Serial
exit status 1
'SoftwareSerial' does not name a type; did you mean 'HardwareSerial'?

Any ideas how to fix it ? Kindly check the softserial example.

Thank you very much

Regards

SnijderC, 2022-11-22:
Hey, can you share the board make and model? SoftSerial is only available on Arduino, and not only that, specific boards have specific ports of Arduino. E.g. the underlying hardware and peripherals are not the same between an Atmega based board vs ESP32, or STM32.. You could be using a board for which there simply isn't a SoftSerial implementation.

PaulMPapa, 2022-11-26:
Hi, thanks for replying. I am using an arduino nano.

vinay-khare, 2023-02-13:
Hi, I am also facing the same issue. I tried on the Arduino Nano as well as the Uno both are not compiling the examples provided in this library. However the SoftwareSerial example from the default arduino library examples are working. Would appreciate any pointers to solve this. Thanks.

Wilhelm1Daniel, 2023-02-13:
Very simple failure...

You have to include the SoftwareSerial.h first, then the DYPlayer...

Now it works for me

silvan2468, 2023-03-05:
Hy
I am also trying to use softwareSerial on an Arduino nano. What have I exactly to do and where/how can I define the specific output pin?
Thanks a lot
Silvan