SnijderC/dyplayer #52: compilation error on AI Thinker ESP32-CAM: undefined reference to `DY::Player::Player(int, unsigned char, unsigned char)'

lionaid, 2023-05-07:
Hi there,

thank you very much for this great work of implementing the communication protocol.

I just tried to use your library with the [AI Thinker ESP32-CAM board](https://docs.ai-thinker.com/en/esp32-cam) and the [Espressif Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) in the Arduino IDE 2.1.0 on macOS 12.3.1.

However, a minimal example (simplified from your esp32 example), using only the creation of a `player` variable with your `DY::Player` class (see below) already fails with this error message: 

```
/Users/lion/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /private/var/folders/q1/jhtqw8ys3kdb_dn83_55fstr0000gn/T/arduino/sketches/0E70F041FA60B7BAC5DE78818341E610/sketch/DY_minimal_example.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_player+0x0): undefined reference to `DY::Player::Player(int, unsigned char, unsigned char)'
/Users/lion/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /private/var/folders/q1/jhtqw8ys3kdb_dn83_55fstr0000gn/T/arduino/sketches/0E70F041FA60B7BAC5DE78818341E610/sketch/DY_minimal_example.ino.cpp.o: in function `_GLOBAL__sub_I_player':
/Users/lion/Documents/Arduino/DY_minimal_example/DY_minimal_example.ino:7: undefined reference to `DY::Player::Player(int, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

DY_minimal_example.ino
```cpp
#include <DYPlayerESP32.h>

DY::Player player(UART_NUM_2, 12, 13);

void setup() {
  player.setVolume(15); // 50% Volume
}

void loop() {
  /* Nothing to do.. */
  delay(5000);
}
```

I would be super happy if you could give me any hints on what the problem might be. 

Cheers!
Lion

alastaira, 2023-08-02:
I think the confusion comes from the fact that the "Arduino" and "ESP32" examples that come with the library don't refer to whether you're targetting Arduino or ESP32 _hardware_, but rather whether you're using the Arduino IDE or ESP-IDF _development environments_.
If you're writing code for an ESP32 board but using the Arduino IDE, you'd still use the "DYPlayerArduino.h" library, as follows:

```
#include "DYPlayerArduino.h"
DY::Player player(&Serial2);

void setup() {
  // Protocol, Rx, Tx pins
  Serial2.begin(9600, SERIAL_8N1, 16, 17);
  player.playSpecified(1);
}
```

lionaid, 2023-10-13:
oh great. I totally didn't get that 😃

closed 2024-02-08 by SnijderC