SnijderC/dyplayer #53: One_Line-Mode solved

silvan2468, 2023-05-08:
Dear all
Because of missing UART-channels, I wrote and tested the one line operation. Not yet all functions, but thats easy to add with my tested functions.
Because I am a beginner in programming and only in C, I don't know how to commit etc.
So perhaps a better programmer than me can/will do it?

To know about one line:
- send lowest bit of a byte first
- timing values written in the table of one-line seams to be not correct. We have to send bits slower. I took 1/3 ms instead of 0,4/1,2 ms. Otherwise it worked not safe.

`const byte oneLinePin = 8;

void send(bool bit) {
  if(bit == 0){
    digitalWrite(oneLinePin, HIGH);
    delay(1);
    digitalWrite(oneLinePin, LOW);
    delay(3);
  }

  if(bit == 1){
    digitalWrite(oneLinePin, HIGH);
    delay(3);
    digitalWrite(oneLinePin, LOW);
    delay(1);
  }
}

void setVolume(uint8_t vol){
  if(vol <= 9){
    const byte data = vol;       // volume 0-9
    digitalWrite(oneLinePin, LOW);   // start
    delay(5);
    for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
    digitalWrite(oneLinePin, HIGH);   // stop
    delay(10);
  }

  if(vol > 9 && vol <= 30){                    // volume 0-30
    const byte data = vol/10;       // zehner
    digitalWrite(oneLinePin, LOW);   // start
    delay(5);
    for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
    digitalWrite(oneLinePin, HIGH);   // stop
    delay(10);

    const byte data2 = vol%10;       // einer
    digitalWrite(oneLinePin, LOW);   // start
    delay(5);
    for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data2 & bit);}
    digitalWrite(oneLinePin, HIGH);   // stop
    delay(10);
  }

  if(vol > 30){
    Serial.println("Volume max. 30");
    
    const byte data = 3;              // zehner
    digitalWrite(oneLinePin, LOW);   // start
    delay(5);
    for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
    digitalWrite(oneLinePin, HIGH);   // stop
    delay(10);

    const byte data2 = 1;             // einer
    digitalWrite(oneLinePin, LOW);   // start
    delay(5);
    for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data2 & bit);}
    digitalWrite(oneLinePin, HIGH);   // stop
    delay(10);
  }
  
  const byte data3 = 0x0C;          // set Volume
  digitalWrite(oneLinePin, LOW);    // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data3 & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void eqSettingDYHV20T(uint8_t eq){       // set Equalizer Mode 0-4: 0 = normal / 1 = Pop / 2 = Rock / 3 = Jazz / 4 = classic
  const byte data = eq;            // equalizer mode
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);

  const byte data2 = 0x0D;         // set EQ
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data2 & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void loopModeDYHV20T(uint8_t loopMode){       // set Loop Mode 0-7: 0 = play all songs / 1 = repeat the same song / 2 = play one song and stop / 3 = random / 4 = ...
  const byte data = loopMode;      // equalizer mode
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);

  const byte data2 = 0x0E;         // set loop mode
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data2 & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void playSongDYHV20T(uint8_t song){       // select song and play
  const byte data = song;       // song nbr
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);

  const byte data2 = 0x0B;       // confirm song number
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data2 & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);

  const byte data3 = 0x11;       // play
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data3 & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void setDeviceSDcard(){
  const byte data = 0x18;       // sd card selection 0x18 = 00011000 ist nötig!
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void setSystemSleep(){
  const byte data = 0x1B;       // sd card selection 0x18 = 00011000 ist nötig!
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void stopDYHV20T(){
  const byte data = 0x13;          // stop playing
  digitalWrite(oneLinePin, LOW);   // start
  delay(5);
  for (byte bit = 0x01; bit != 0 ; bit = bit << 1){send(data & bit);}
  digitalWrite(oneLinePin, HIGH);   // stop
  delay(10);
}

void initDYHV20T(){
  pinMode (oneLinePin, OUTPUT);
  digitalWrite(oneLinePin, HIGH);
  delay(100);
  setDeviceSDcard();
  setVolume(20);
}`

I hope my code is helpful for someone else.

Greetings
Silvan

paulklinkenberg, 2024-07-20:
Hi Silvan,
Thanks for sharing your code! It works with the DY-SV17F and Arduino Nano.
The only thing bothering me, is that after calling setSystemSleep(), there is no change in power usage :(
Did you have more luck when testing this functionality?

silvan2468, 2024-07-20:
No. As I needed a realtime system (no delays), at the end I realized, that its possible with Arduino Nano 33 Ble (NRF52840) to have 2 serial ports. Not just one. So I changed back to true UART  communication and did not longer need the one-line-mode.

paulklinkenberg, 2024-07-20:
Aah, that makes sense. Thanks.
I was actually also using UART, but in my battery driven project, I noticed the battery was being drained too fast. I was hoping for the “system sleep” functionality to help me, but I noticed no difference.
Do you (or anyone else) know if the “system sleep” mode is available via uart? I couldn’t find it in the docs, not in the source code of this repo…