SnijderC/dyplayer #54: Impossible to create Player with SoftwareSerial

Alexandruino, 2023-07-27:
When trying to execute the SoftwareSerial.h example, I get the following error:

SoftSerial:7:30: error: no matching function for call to 'DY::Player::Player(SoftwareSerial*)'
 DY::Player player(&SoftSerial);
                              ^
/home/alex/Documents/RizomeAudio/SoftwareSerial/SoftSerial/SoftSerial.ino:7:30: note: candidates are:
In file included from /home/alex/Documents/RizomeAudio/SoftwareSerial/SoftSerial/SoftSerial.ino:3:0:
/home/alex/snap/arduino/current/Arduino/libraries/dyplayer-main/src/DYPlayerArduino.h:29:5: note: DY::Player::Player()
     Player();
     ^
/home/alex/snap/arduino/current/Arduino/libraries/dyplayer-main/src/DYPlayerArduino.h:29:5: note:   candidate expects 0 arguments, 1 provided
/home/alex/snap/arduino/current/Arduino/libraries/dyplayer-main/src/DYPlayerArduino.h:24:9: note: constexpr DY::Player::Player(const DY::Player&)
   class Player : public DYPlayer
         ^
/home/alex/snap/arduino/current/Arduino/libraries/dyplayer-main/src/DYPlayerArduino.h:24:9: note:   no known conversion for argument 1 from 'SoftwareSerial*' to 'const DY::Player&'
/home/alex/snap/arduino/current/Arduino/libraries/dyplayer-main/src/DYPlayerArduino.h:24:9: note: constexpr DY::Player::Player(DY::Player&&)
/home/alex/snap/arduino/current/Arduino/libraries/dyplayer-main/src/DYPlayerArduino.h:24:9: note:   no known conversion for argument 1 from 'SoftwareSerial*' to 'DY::Player&&'
exit status 1
no matching function for call to 'DY::Player::Player(SoftwareSerial*)'

I am out of ideas on how to fix this. Could you please help me?

Thank you in advance.

Alex.

Jeff-Haas, 2023-08-05:
Try the last comment in this other issue:
https://github.com/SnijderC/dyplayer/issues/49

alastaira, 2024-07-02:
Are you deploying to a target platform that actually supports SoftwareSerial? How are you initialising the SoftSerial instance?