SnijderC/dyplayer #59: ESP32 Support

ConnalM, 2024-02-25:
Hi,

I'm looking to use your library to connect an ESP32 to a DY-SV5W or similar but when I use the Arduino header file, I hear nothing and, when I use the library for the ESP32, I get an error:

error: no matching function for call to 'DY::Player::Player()' DY::Player player;

Any suggestions?

Thanks,

Connal

alastaira, 2024-04-11:
ESP32 works fine for me, prorgammed using Arduino IDE.

darrenjcosborne, 2024-06-11:
@alastaira Can you please share your working code?

alastaira, 2024-07-02:
It's basically exactly the same as in the examples. Relevant lines as follows:

```
// INCLUDES
// Used to create serial interface to DY-HV20T (DY-SV8F/DY-SV5W/similar)
// See https://github.com/SnijderC/dyplayer
#include "src/DYPlayerArduino.h"

// GLOBALS
// An object to address the audio player
DY::Player player(&Serial2);

void setup() {
  // Initialise secondary serial interface to audio module
  Serial.println(F("Configuring audio interface"));
  Serial2.begin(9600, SERIAL_8N1, 16, 17);
  player.begin();
  player.setCycleMode(DY::PlayMode::OneOff);
  // Volume is a value from 0-30
  player.setVolume(30);
}

void loop() {
  player.playSpecified(2);
}
```